robertzach/ewinch_remote_controller #3: Motor

looxis-renaud, 2022-09-18:
In one of the youtube comments you mentioned that you are using the QS Motor 12kw,
available here: https://www.aliexpress.com/item/4001237537247.html

Do you use the **"72V"** or the **"96V"** version ?

robertzach, 2022-10-17:
Chinese supplier are a bit confusing with their order options.
The main thing is the "KV" value which describes the rpm / voltage relation.
rpm = KV * voltage

Therefore, it depends on what battery voltage you use.
With a 16s lion battery (what I use) you have around 16*3,7 --> 60V.
For this case I recommend the 72V option (72V @ 1047RPM --> KV = 14).

closed 2022-10-17 by robertzach

steve1428, 2023-04-05:
be careful, I Found some other people representing themselves as QS motor when they weren't.  Deal with them directly.