robertzach/ewinch_remote_controller #5: Which firmware for VESC Controller?

looxis-renaud, 2022-09-18:
VESC is controlling the brushless DC motor - do you use the standard VESC Firmware
or a fork? Asking this question because in one of your youtube videos you mention "Skypuff" which seems to be a fork of VESC standard firmware. I found this: https://github.com/talkyoo/vesc_tool_skypuff

robertzach, 2022-10-17:
It uses standard Vesc. e.g. 5.3.
Autostop is only supported with some changes in the standard vesc code:
see vesc/vesc_ppm_auto_stop.patch

looxis-renaud, 2022-10-18:
Hello Robert, I assume I could just use /vesc/vesc_75_300_auto_stop.bin,
which is probably the newly compiled VESC firmware including your vesc/vesc_ppm_auto_stop.patch ?

I do have the VESC 75/300.

_______
yes, I do not have any coding skills, I am just able to read and possibly understand the most basic code...