robertzach/ewinch_remote_controller #9: vesc_75_300_auto_stop.bin makes the motor spin without stopping

looxis-renaud, 2023-03-12:
Hello Robert,
I had no luck flashing this firmware onto the VESC 75/300.
once you start setting up the input with this firmware, the motor just starts spinning and isn't controllable with your remote.
It does work with the original firmware, though (just no autostop then...)

I assume this is because of this function:

```
        // my default pull
        // manualy regulate pull with a poti on ADC2
        // in manual pull mode autostop only on demand (to rewind the last meters cable...)
   #ifdef ADC_IND_EXT2
     if ( (((float)ADC_Value[ADC_IND_EXT2])/4095) * V_REG > 0.5) {
     my_default_pull = ((((float)ADC_Value[ADC_IND_EXT2])/4095) * V_REG - 0.5) * 30;	//0 - 100% after 0.5V 
        // set default pull value
     mc_interface_set_current(my_default_pull * 0.35);	//0 - 35A
```

This sounds like as it would be an early version of your code, where you were using a potentiometer to test the motor?
ADC2 input is not used, it's the PPM input that's being used... which has no ability to control the motor with this firmware.

I am just speculating here... sorry to bother.

robertzach, 2023-03-12:
I have a poti on ADC2 to manually control the winch. E.g. To wind up the last meters of the line when finishing. Or to set a tension when used as a rewind winch...

If you do not install one, just connect ADC2 to GND.

closed 2023-03-12 by robertzach

looxis-renaud, 2023-03-27:
@robertzach , can you tell me what specifications your potentiometer has ? i.e. the resistance, like 0,1, 1, 10 or even 100 kOhm ?