robertzach/ewinch_remote_controller #17: VESC: Motor is "pumping" under heavy load, is this normal?

looxis-renaud, 2023-05-13:
Sorry Robert to bother you again. 
I am ready to make a test flight, but we're having an issue: 
 we wanted to test the motor's drag force according to the defined force in the remote, before we are actually going to launch ourselves. 

We attached a "Quad" (a motorcycle with 4 wheels) to the line and started pulling. To measure the drag force in kilogram (or daN) we were using heavily the brake of the quad. 

As you can see in this video: https://youtu.be/rusCd4p7q5k
the Motor is pumping /oscillating heavily under load.

When driving backwards under load, the winch was nicely unwinding the line, no issues with oscillation.

Is this possibly because of the inertia of the quad and because of the resistance of the wheels on the bumby grass? 

Have you not experienced something similar? 
Could this possibly be solved with some settings in the VESC app? 

Thank you in advance for any help.

robertzach, 2023-05-15:
Looks like your VESC motor configuration is not very stable.
You can compare the pid values with the following config: https://github.com/robertzach/ewinch_remote_controller/blob/main/vesc/vesc_motor_config_12kw_260_V4.xml
Furthermore it is import that your winch is mounted on a more solid point to avoid oscillation!!
You can call me at +43 664 2314337, I may can give you some hints on the VESC motor config.
PS: Please also let me know were you got your thread for the line guide from? It looks really nice!

looxis-renaud, 2023-05-16:
Hi Robert,
thanks again, as always, for your valuable feedback. will check the VESC settings again and also give you a call.

The "line guide" or "winding mechanism" was purchased from https://www.ebinger-gmbh.com/
It's the model 1.610.010 >>automatische Schlauchführung, Stahl lackiert, 1/2" -/ 5/8" Schlauch für Modell OSM 610<<.
It's intended purpose is for water hoses or compressed air hoses and was modified by a friend of mine to accommodate for the thin dyneema line and the width of the drum.

closed 2023-05-16 by looxis-renaud