robertzach/ewinch_remote_controller #19: Skalierung der Leistung

Logo66, 2023-07-29:
Hallo Robert Microkontroler programmieren usw. ist Neuland für mich. Habe eigentlich fast alles hingekriegt aber das eine oder andere versteh ich noch nicht ganz. z.B
In der Anweisung für PPM-Einstellungen heißt es: „static int myMaxPull = 85; // 0 – 127 [kg], muss mit den VESC-ppm-Einstellungen skaliert werden.“ Wie wird dies eingestellt oder kalibriert?

https://photos.app.goo.gl/PnyRyFPGkBvydZn87

steve1428, 2023-07-29:
I think he means you just vary the pull amount in the code depending on
your particular motor's power
if you notice he's made two of them so he can use the other one in braking
mode to test and set things up, otherwise you have to go out in a field and
do test pulls.
I've got mine all working I just need to 3D print the hand held controller
case because I can't find anything appropriate off the shelf
https://youtu.be/9rqX89-msBY

Logo66, 2023-07-29:
https://a360.co/3Dy5PxJ

Ich habe mir diesen gedruckt.

steve1428, 2023-07-29:
Das ist wunderschön !

steve1428, 2023-07-29:
https://d-splicer.com/d-splicer-set-kit/

robertzach, 2023-07-29:
Depending on your motor (power, KV value, diameter, etc.) you need to scale the resulting kg pull on your line.
I think the best way is to adapt the "Motor Current Max" and "Motor Current Max Brake" value in the Vesc Motor Settings (General -> Current).
You could also adapt the "Pulselength Start" and "Pulselength End" in the Vesc App settings (PPM -> Mapping) to reach the same goal.
I used a suitcase scale to measure the real pull on the line.
With my QS260 hub motor I have around ~3,6A/kg pull.

closed 2023-07-29 by robertzach

Logo66, 2023-07-29:
Danke Robert das hat mir weiter geholfen.