robertzach/ewinch_remote_controller #20: QS motor

lamaatje, 2023-10-13:
Hi Robert,

In understand you use a 12kW QS motor 13" by 4.5" is that correct?
Did you order a specific variant optimized for torque?

Kind regards,
Stefan

robertzach, 2023-10-16:
I used https://de.aliexpress.com/item/4001237537247.html?gatewayAdapt=glo2deu
with standard config.
I reach about 60 km/h with my 16s battery and a pull force about 70kg with about 240A.

steve1428, 2023-10-16:
I shorted my trampa vesc trying to setup on my electric unicycle, we tried
solder new mosfets but didn't work.
There is cheaper vesc from India now

lamaatje, 2023-10-16:
> Links:

Hi Robert,

Thanks for the response. I am looking for an appropriate motor it is my 
understanding that many of those hub motors used in scooter/motorbike 
applications are optimized for speed.
So am looking for something with high Kt [Nm/A] and high rated current. 
I have a second hand QS motor that just does that but it does not have a 
type tag. I can pull 70kg with 100A with a radius of .25m
this means 700/4= 125Nm @ 100A -> Kt = 1,25 Nm/A.
I am not sure what you diameter is but lets also take .25m than your 
motors Kt = 125/240 = 0,52 Nm/A.  Since motor dissipation is mainly due 
to copper loss I2R, lower current is very desirable to avoid motor 
heating. Of course with low R heating can also be avoided that is why it 
is important to know the rated current. For your motor I understand that 
this is 167A so that means it will heat up considerable.
Can you confirm this?

Kind regards,

Stefan

steve1428, 2023-10-16:
yup, hub motors aren't really designed for high torque applications so they
get hot.
setup a fan and keep an eye on the temps.
If you look at other winch projects on youtube some guys are using other
motor types.

lamaatje, 2023-10-16:
Well there is not a fundamental reason why they could not be optimized for torque and since they have high pole pairs counts they can deliver quite a lot of power in the RPM range for a paraglider winch applications. Although the high rotational inertia is also a big disadvantage. The big advantage is direct drive. Do you have some pointers wrt other QS motor or hub motors in general?

steve1428, 2023-10-16:
yes they do make very high torque BLDC motors..... the leperkim veteran
patton electric unicycle (126V) I just bought has enough torque to spin the
wheel when you climb a hill, it's a beast.

robertzach, 2023-10-16:
**If you want to do step tows you need some speed to be safe when flying back!!!**
If you do no step tows you can go down with the speed to around 40km/h -> choose a hub motor with lower KV value (more torque optimized)
My first winch was calculated for max speed around 40 km/h (see https://www.youtube.com/watch?v=E3TgccQ8CuA).
It also depends on what battery you want to use.
Right now I am using 16s which is 66V on full charge. This keep my battery setup easy.
For my first winch I increased to 22s (90V) to get more speed.
To sum up, battery/controller/motor needs to be chosen in the right mix, depending on your speed (step tows?) and pull (weight?) requirements.

lamaatje, 2023-10-16:
Hi Robert,
I have to disagree. When step towing and flying away from the winch the motor just needs to brake. The torque is opposite to the rotation direction so you will be generating power. E.g. say you fly away from the winch with 15 m/s and the tow force is 40N (towards the winch ie the same direction as when doing a tow) you will be generating 600W. Assuming no losses for simplicity, if your battery is 60V you will be charging it with 10A. 

Towing requirements are simple in zero wind conditions 10 m/s @ 1000N. So 10kW mechanical power.  So a 12 kW rated motor should be able to do the job. 12kW is the electrical rating assuming 10/12=83% efficiency at the rated mechanical power this is realistic. Depending on your spool radius you need a motor with the proper Torque, speed curve where the RPM at maximum power is at 10 m/s. For a radius of 0.2m that would be 477 RPM and 200Nm.

steve1428, 2023-10-16:
12KW is plenty of power regardless

I've got the QS motor 3/5 kw motor and I know people are able to get it to
work

robertzach, 2023-10-17:
I recommend you https://en.wikipedia.org/wiki/Motor_constants
Max speed is defined by KV = speed / Voltage.
The only way to increase speed above the limit based on KV and Voltage is to use field weakening.
But field weakening is still beta in VESC and in my tests it only increased the max. speed with my hub motors around 20%.
I would not recommend to used field weakening to increase max. speed for flying away on step tows.
If anything fails you can get in big trouble....

lamaatje, 2023-10-17:
I think you are missing the point here. All these things like field weakening to increase speed are applicable when the motor is **driving**. When you fly away from the winch the motor is not supposed to be driving it is supposed to be **breaking**.  If you look to it from another perspective: suppose that you completely shut down the motor control when you fly away from the winch will your speed be limited? I think the answer is no. Is it is also not desirable because you need some force to limit slack of the line and make sure the line stays on the spool. You can do this by letting the motor brake a bit this can be done at rpm's way higher than the rpm's in the range were the motor is driven. You have to basically control the current the motor is **generating** (NB not consuming) and the best way is to have the current back in the battery (regen braking). The speed range were you can do regen braking is much higher and mainly voltage limited. I have this working on my controller and it works just fine.

lamaatje, 2023-10-17:
> 12KW is plenty of power regardless I've got the QS motor 3/5 kw motor and I know people are able to get it to work
> […](#)
Sure, with a little bit of head wind, low pilot weight, low ascend rate. But that is not the point. If your motor is capable of a certain sustainable mechanical power you want to dimension it in such a way that it gives that output in a range that is optimal for towing that's why you do the math.

robertzach, 2023-10-17:
> when you fly away from the winch will your speed be limited? I think the answer is no

This is wrong. For brushless motor rpm and voltage are in direct correlation.
Therefore, when you increase the motor rpm (when flying away) above the limit (which is the voltage of the controller) the controller will burn. By using field weakening you can push this limit a bit, but any issue will immediately burn the controller.
When you burn the controller when flying away I wish you that the lines on the spool do not get messed up.
Yes, it is possible to a 5kW hub motor, but you will be very likely limited with speed supported...

steve1428, 2023-10-17:
doesn't the BMS deal with the regen current?  This is a problem with
electric unicycles going down hill after a full charge.
as long as the battery isn't 100% the regen will go to the battery no?

robertzach, 2023-10-17:
The hardware diagram is:
motor ---  controller --- battery (incl. BMS)
If the controller can't deal with the voltage of the motor you don't have to worry about the battery.
The controller will be dead in a few seconds anyway.

closed 2023-10-17 by robertzach

lamaatje, 2023-10-17:
*This is wrong. 
No there is no force holding you back.

*For brushless motor rpm and voltage are in direct correlation. 
That is correct that is why I state "The speed range were you can do regen braking is much higher and mainly voltage limited." so I agree that your controller has to be able to deal with the voltages but that is a **controller property**

*By using field weakening you can push this limit a bit, but any issue will immediately burn the controller. When you burn the *controller when flying away I wish you that the lines on the spool do not get messed up. 
No field weakening has nothing to do with it. Yes if voltages are higher than the MOSFET Vds breakdown voltage that is a problem but it is controller related. I have a motor when driving it at zero load reaches 634 RPM the voltage is than around 70V that is a **motor property**. The torque is than long gone and my tow line speed is than around 55 kmh. So when driving the motor the RPM is limited because the back-EMF reaches input voltage of 70V. With field weakening you actively can reduce the back-EMF and gain a little speed or you can increase the voltage when driving the motor. My controller can handle 120V so when I fly back in windy conditions with 70 kmh the motor will generate around 90V and there is not issue whatsoever to do controlled EM braking.
So when choosing your motor's Kv the step tow fly back speed is not a real consideration just make sure you have a controller that can deal with the generated voltage.

robertzach, 2023-10-21:
I am interested in your controller which can handle 120V.
Which one is it?

steve1428, 2023-10-21:
are you asking me because I mentioned my 126V veteran patton EUC?

it's the controller the mfg put on it.

The 3SHUL makes a controller for that voltage, you can see their products
here

https://alienrides.com/collections/3shul-vesc-motor-controllers/products/3shul-vesc-motor-controller-c1400

lamaatje, 2023-10-21:
Hi Robert,

It is a sevcon gen4 96-120V. I included an overview pdf. The 72-80V 
variants already support up to 116V DC maximum operating voltage so that 
would be also a good choice in this example.
I also have a question to you wrt the QS motor you purchased was it with 
rim? I understand it is also possible to order them without rim in that 
case the minimum spool diameter can be a bit smaller which gives more 
tuning bandwith to match the motor torque speed curve.

steve1428, 2023-10-22:
I would warn people away from buying the Trampa VESC, it's designed to fit
in an electric skateboard, very compact and delicate, surface mount mosfet,
microscopic gate drivers and associated components positioned close by,
it's next to impossible to repair and easily damaged.

robertzach, 2023-10-22:
3SHUL VESC controller look really good!
Do you have experience with the CL350?
(https://alienrides.com/collections/3shul-vesc-motor-controllers/products/3shul-vesc-motor-controller-cl350)

steve1428, 2023-10-22:
no but check out Luke F youtube channel, he is using one for his winch,
i'll post the link to his video but I don't know if github allows.
if not look for:
"Playing music on the hang glider winches motor stator"

https://youtu.be/fslWr41YOQk?feature=shared

lamaatje, 2023-10-22:
I saw there is also a European dealer but the 350CL is only on presale. Very affordable compared a commercial controller. Does VESC have an API or protocol to write application?

(https://alienrides.com/collections/3shul-vesc-motor-controllers/products/3shul-vesc-motor-controller-cl350)

steve1428, 2023-10-22:
i dunno bout this expensive high voltage dc stuff, check out what this guy
is doing with the GY6 drivetrain

https://youtu.be/VPotT-ulV7g?feature=shared